DustinWatts/FreeTouchDeck #1: Suggestion: Provide example config files for TFT_ESPI

witnessmenow, 2020-10-24:
I think it would be a good idea to provide example config files for TFT_ESPI, especially for the way you recommend connecting everything up. When I get the makerfabs board working I'll share my one for that

DustinWatts, 2020-10-24:
I agree! Although TFT_eSPI has some good documentation on that, dragging a file known to be working is a lot easier!

DustinWatts, 2020-10-25:
Created a folder called `user_setup.h Examples` and uploaded my user_setup.h. I'm still doubting whether to have lot's of sub directories and in each of those a `user_setup.h` file. Or to do it like it is now, one directory and naming the setup files to give the best description but then have the user renaming the file to `user_setup.h` again. Off course copying the contents would also work.

DustinWatts, 2020-10-25:
See: https://github.com/DustinWatts/FreeTouchDeck/tree/master/user_setup.h%20Examples

DustinWatts, 2020-10-25:
Feel free to make a Pull Request with your User_Setup.h configuration.