DustinWatts/FreeTouchDeck #6: Allow for more steps per keypress

witnessmenow, 2020-11-01:
Currently it is limited to 3.

Might require some rework, but I think using an array to store the steps would allow people to do more complicated actions.

DustinWatts, 2020-11-01:
I think switching from a static situation to an array with the actions would make this a lot easier! Work in progress! :)

Lucero-delcielo, 2021-05-08:
Is there a way for a button to type a string of characters, say a web address, rather than just a few keys?  I'm wondering if there is a way to write this directly into the code because I don't see how to do it in the configurator. Thanks!

Lucero-delcielo, 2021-05-26:
For now, I am running Autohotkey scripts, and linking those to my button on the FTD. This works well, but is definitely not how I want to do it permanently. Thanks for the ideas and support, this is a great project!

DustinWatts, 2021-05-26:
AHK is very handy indeed. I use it a lot. Can you tell me what exactly doesn't work? And you are correct I forgot the `;` in my comment. I did notice that there is a `}` missing, try this:

```
else if (b == 2) // Button 2
{
  const char string[] = "gmail.com";
  for(int i=0; i < strlen(string); i++ ){
   char c = string[i];
   bleKeyboard.print(c);
   delay(10); // maybe increase this a bit
 } // <- was missing
 if (menu5.button2.latch)
 {
  if (islatched[22])
  {
   islatched[22] = 0;
  }
  else
  {
   islatched[22] = 1;
  }
 }
}
```

I can verify that this works. But keep in mind that in the code "button 2", is the third button on your FreeTouchDeck.

newwolf, 2021-11-08:
Any change this will be implemented in a next release as default option?
It's a feature I really like.
The panel is now laying at the corner of my desk and crying for getting used at the moment . ;-)

danielo515, 2021-11-15:
A function to include custom user actions have been included. It requires a bit of code on your side, but you will be able to then configure from the GUI.
Will that work for you @newwolf ?

closed 2022-06-15 by github-actions[bot]