DustinWatts/FreeTouchDeck #7: Configurator web interface is slow to load

witnessmenow, 2020-11-01:
The web interface for the configurator takes about 20 seconds to load for me (didnt time, but it's slow)

Any idea what is causing this?

I'm using Firefox on Windows 10.

DustinWatts, 2020-11-01:
This is something that is known to be an ESP32 thing. I'm trying out some alternatives to the WebServer library I'm using now. My experience is that the more requests you send to the ESP32 the slower it gets.

DustinWatts, 2020-11-07:
09bc3dc1379095a75623b51246f3c00995af1fde Makes the switch to ESPAsyncWebServer. Although this requires the user to install two additional libraries, I think the speed gained by this is very much worth it. If you experience any speed/reliability issues, feel free to re-open.

closed 2020-11-07 by DustinWatts