DustinWatts/FreeTouchDeck #8: Display version number somewhere

witnessmenow, 2020-11-01:
Would be good to have a code version number displayed somewhere so people could mention what version they are on in issues

DustinWatts, 2020-11-01:
It displays it during boot in the top left corner. But after there is no way as of yet. I am planning the one still unused button in the "Settings" menu an "Info" button. Displaying all kinds of information like version, BLE Name/MAC, battery voltage (if a battery is used) and some other stuff. If you have other ideas what can be on that page, let me know!

DustinWatts, 2020-11-04:
Commit https://github.com/DustinWatts/FreeTouchDeck/commit/6e6fffe14f2e160d4251ad7cb5852b543e2ead14 added an info screen to the settings menu. Displaying the FreeTouchDeck version and some other useful information.

closed 2020-11-05 by DustinWatts