DustinWatts/FreeTouchDeck #13: FreeTouchDeck for Home Assistant

CeJienAJPC, 2020-11-22:
Is there a way, for you to bring this awesome project to Home Assistant too?
I already commented on your Youtube Video regarding this (repeating here, coz this is a proper channel for adding NEW REQUESTS). Apologies though.

FreeTouchDeck + Home Assistant = Too many interesting use cases

Would love to see this project with Home Assistant

Regards

DustinWatts, 2020-11-22:
I've been looking in to this and Home Assistant exposes an RESTful API. I'm not sure yet on how to implement the use of this API (I mean the user interface and configuration), but I have been testing using HTTPClient at the same time as Bluetooth and those seem work together just fine.

Docs: https://developers.home-assistant.io/docs/api/rest

DustinWatts, 2020-12-28:
Well... did some work on it the past month. And some basic integration is working now in my development version that I have locally. The part that doesn't work is is when HA is configured with DuckDNS to use SSL because there is not enough ram on the ESP available to start a secure network client. However, HA can be configured to use both SSL for outside connections and normal HTTP over internal connections. So I'm probably going to release this version with HTTP HA support together with a tutorial on how to set it up. 

But... progress is being made!

CeJienAJPC, 2020-12-29:
That's a great progress. Thanks Dustin

Syndlex, 2021-02-08:
@CeJienAJPC 
Hmm I really like the Idea of Home Assistant integration on TouchDeck but there are already Projekt that are Implementing this and are Home Assistant centered. 
For example: https://esphome.io/components/binary_sensor/nextion.html

I thing a better way would be to keep the focus of FreeTouchDeck in the core functionality and extensibility of the project.
This would free up resources here. 
 
 Note: Giving Free Touch Deck an extensible component structure could help implement different Use Cases more easily.

romix123, 2021-02-15:
I was thinking of expanding the code to allow for mqtt messages to be sent. Should not be hard to implement. Action 11 is available. Duplicating the wifi struct for mqtt server and adding config action field for mqtt messages. Once an mqtt message is set/reset, a counter should be updated allowing the device to determine whether connection to the broker needs to be established on boot.
I can’t see how you responded to the HA request, though.

DustinWatts, 2021-02-15:
@romix123 One of the issues is that a combination of the TFT library, BLE, WiFi and the MQTT pubsubclient takes up a lot of resources. So a combination of all these functionalities is quite a challenge. My response to the HA request was that I have a version given to a few people on my discord to test. But if you'd like to help and get a stable version with MQTT, I'm up for it!

romix123, 2021-02-16:
I will give it a shot. Browsing through the code, it feels like there is room for code optimisation as well. 

Best, 
Ronald Leenes

Verstuurd vanaf mijn iPad

DustinWatts, 2021-02-16:
@romix123 I totally agree with there being room for code optimization!

Groet,
Dustin

gjelsoe, 2021-05-31:
Take a look openHASP, it runs on Dustins hardware.

seraasch, 2021-06-12:
I'm going to vote for adding OpenHab to the list... There is a REST interface as well as MQTT.

rlust, 2021-07-22:
> Take a look openHASP, it runs on Dustins hardware.

I want to try this on openhasp! What binary does the ESP32 use? Please share if you have this working!

DustinWatts, 2021-07-23:
> > Take a look openHASP, it runs on Dustins hardware.
> 
> I want to try this on openhasp! What binary does the ESP32 use? Please share if you have this working!

You can find the binary in the release: https://github.com/HASwitchPlate/openHASP/releases/tag/v0.6.0

rlust, 2021-07-24:
Which binary do I use for a ESP32 WROOM DEV module?

DustinWatts, 2021-07-24:
> **I am getting the above error.**

Hmmm... I'm not sure what's happening there. Can you open an issue here? : https://github.com/HASwitchPlate/openHASP/issues

rlust, 2021-07-25:
Thanks for the help! I got openhasp loaded. Do you have any ideas on how to configure OpenHasp to use your TFT Combiner. Have the web page working but no display. I am using a ILI9488.

DustinWatts, 2021-07-26:
You have to define the correct pins to use. The pins are:

TFT_BL   32          
TFT_MISO 19
TFT_MOSI 23
TFT_SCLK 18
TFT_CS   15
TFT_DC    2
TFT_RST   4 

TOUCH_CS 21

Also, you can try check if the display is working by uploading an example from the TFT_eSPI library. Use the User_Setup.h that is in the examples folder in this repo. Use the "resistive touch" verison.

rlust, 2021-07-26:
Did you have to short J1 pin to make 3.5 TFT SPI ILI9488 work with ESP32 ?

danielo515, 2021-11-15:
So, I want to generalise this issue to, instead of integration with HA, add the ability to send HTTP requests.
That will allow to integrate it with any REST service of your choice. Then we can write guides about how to integrate HA if that is what you want using the REST API, along with other services.

By the way, a mention to the openhasp project with a note on which binaries to use will also help a lot of people find the right resources.

frostworx, 2022-06-07:
_Preventing the bot from closing the issue_ _(at least I think this comment stops it 😀)_

frostworx, 2022-08-07:
:)

DustinWatts, 2022-08-28:
Should be possible soon with the new version of FreeTouchDeck and in combination with an ESP32-S3

closed 2022-11-04 by github-actions[bot]