DustinWatts/FreeTouchDeck #18: Add reference to REPEAT_CAL to installation wiki

R0CKB0TT0M, 2020-12-17:
after following the installation instructions to the . I had a problem where it wouldn't take any of my touch inputs. turns out all I had to do was calibrate it and it does say so at the top of the .ino file but it would be nice if the instructions also mentioned it.
Also would it be possible to make it receive commands over serial to start calibration or connect to WiFi.

DustinWatts, 2020-12-18:
Good point. I will add this to the wiki.

Regarding the possibility to send serial commands to FreeTouchDeck that should be possible.

DustinWatts, 2020-12-18:
Something like this in the loop() would work for re-calibrating without needing to re-upload the sketch:

```
if(Serial.available()){
      
        String command = Serial.readStringUntil('\n');
 
        if(command == "CAL")
        {
          FILESYSTEM.remove(CALIBRATION_FILE);
          ESP.restart();
        }
    }
```

Regarding connecting to WiFI. Do you mean setting your credentials? Or actually connecting to WiFi?

R0CKB0TT0M, 2020-12-18:
I was talking about actually connecting but having both would be optimal.
amazing work you've done here by the way.

DustinWatts, 2020-12-18:
Thanks! Setting the SSID and password through a serial connection is something that could be very useful. I'm just wondering why do you want to connect to WiFi through the serial monitor?

DustinWatts, 2020-12-26:
https://github.com/DustinWatts/FreeTouchDeck/commit/f054b24a2bf314e6a50a39176e69058a24ebef11 Adds the ability to edit the wifi configuration through a serial connection. Described here: https://github.com/DustinWatts/FreeTouchDeck/wiki/5.-WiFi-and-Internet 

Calibration can be reset by sending `cal reset` though a serial connection.

closed 2020-12-26 by DustinWatts