DustinWatts/FreeTouchDeck #20: Help me

mbazzo, 2020-12-23:
I'm trying to build and is showing me the following errors:

In file included from C:\Users\bazzo\Documents\Arduino\FreeTouchDeck\FreeTouchDeck.ino:257:0:
sketch\Touch.h: In function 'void touch_calibrate()':
Touch.h:37:9: error: 'class TFT_eSPI' has no member named 'setTouch'
     tft.setTouch(calData);
         ^
Touch.h:59:9: error: 'class TFT_eSPI' has no member named 'calibrateTouch'
     tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15);
         ^
C:\Users\bazzo\Documents\Arduino\FreeTouchDeck\FreeTouchDeck.ino: In function 'void loop()':
FreeTouchDeck:523:19: error: 'class TFT_eSPI' has no member named 'getTouch'
     pressed = tft.getTouch(&t_x, &t_y);
                   ^
FreeTouchDeck:599:19: error: 'class TFT_eSPI' has no member named 'getTouch'
     pressed = tft.getTouch(&t_x, &t_y);
                   ^
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\bazzo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
'class TFT_eSPI' has no member named 'setTouch'

mbazzo, 2020-12-23:
sorry, i ended up moving to another folder.

mbazzo, 2020-12-23:
I did the whole procedure but it just returns me this error

17:05:07.901 -> rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:05:07.901 -> configsip: 0, SPIWP:0xee
17:05:07.901 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:05:07.901 -> mode:DIO, clock div:1
17:05:07.901 -> load:0x3fff0018,len:4
17:05:07.901 -> load:0x3fff001c,len:1216
17:05:07.901 -> ho 0 tail 12 room 4
17:05:07.901 -> load:0x40078000,len:9720
17:05:07.901 -> ho 0 tail 12 room 4
17:05:07.935 -> load:0x40080400,len:6352
17:05:07.935 -> entry 0x400806b8

DustinWatts, 2020-12-23:
'class TFT_eSPI' has no member named 'getTouch' is likely because you haven't defined TOUCH_CS in the User_Setup.h file. 

As to the second error. I need a bit more information. Can you tell me what ESP32/Screen combination you use? Also, what are your settings under "Tools" when you upload the code? Does uploading the `data` folder using the ESP32 Sketch Data Upload tool work as expected?

Make sure when flash size is correct when you flash the ESP32.

To rule out a hardware issue, you could try and upload a `blink` sketch and see if that works.

closed 2020-12-25 by mbazzo