DustinWatts/FreeTouchDeck #21: Setting up more subfolders

mbazzo, 2020-12-25:
It would be nice to put settings for more folders (subfolders). so you can put for example:

Games
        Steam
                      Dota 2
                      CS
                      etc....
        Blizzard
                      COD
                      WOW
                      etc....

DustinWatts, 2020-12-30:
That would be very nice indeed. The number of buttons/menus is for now hard coded. I'm working on a soft coded version, but this requires a lot of re-writing because I locked myself in to this structure from the start. But as more flexible button/menu configuration is request quite often it is worth spending time coding!

mbazzo, 2020-12-30:
It would also be good if you could change the background color of the buttons individually.

sle118, 2021-06-03:
@mbazzo still in very early stages, but I'm doing my best to achieve some of the flexible goals you are craving for. Single button background color: check!

However, there's no front end at this point to make configuration easy and no documentation aside from making sense of the code. But all of what you are asking for is there and @DustinWatts has been kind enough to show some interest in my contribution.

So if you have a touchdown to spare for the bleeding edge (might want to consider getting another one to spare your daily driver) see you here
https://github.com/DustinWatts/FreeTouchDeck/issues/51
And here for the yet to be merged fork:
https://github.com/sle118/FreeTouchDeck