DustinWatts/FreeTouchDeck #22: Latching not disappearing completely when using 320x240 display

charliejgallo, 2020-12-31:
everything else works perfectly with an ili9341 (#define ILI9341_DRIVER and #define SPI_FREQUENCY  40000000 changed in user_setup.h and resolution set to 320 x 240 in FreeTouchDeck.ino) same wiring as the ili9488 resistive photo attached.

[image]
[image]

maybe something to do with the spacing and "rounding" of the buttons due to resize??

DustinWatts, 2020-12-31:
That is a good point. The latching dot is a fixed distance from the image on the button. What happens is, when the dot is drawn outside the button boundaries that part is not redrawn when "unlatching". 

In `drawhelper.h` on line 16 the latching dot is drawn: 
`  tft.fillRoundRect((KEY_X - 37 + col * (KEY_W + KEY_SPACING_X)) - 12, (KEY_Y - 37 + row * (KEY_H + KEY_SPACING_Y)) - 12, 18, 18, 4, generalconfig.latchedColour);
}`

The number 12 is responsible for the offset. You could change it to a smaller number to bring in the dot closer. I do not have a 320*240 screen but when resize my screen and use 2 instead of 12, the latching dot falls completely within the button. 

Maybe if you can let me know what position works for you. I can change the code to make it a ratio of the screen size. Or just put and if statement to check whether the screen is 480*320 or 320*240.

charliejgallo, 2020-12-31:
`tft.fillRoundRect((KEY_X - 37 + col * (KEY_W + KEY_SPACING_X)) - 2, (KEY_Y - 37 + row * (KEY_H + KEY_SPACING_Y)) - 2, 18, 18, 4, generalconfig.latchedColour);` worked great!!

[image]

also had to set text size to 1 (instead of 2 ) in line 1496 in Drawhelper.h for getting the system information to fit the screen..

everything else works ok with this ili9341 display, even the sleepwake function with the touch interrupt!

thank you and happy new year!!!

DustinWatts, 2021-01-02:
Thanks for that @charliejgallo. In the next update I will include a few lines that checks the size of the display and then decides where to put the latching dot. I will also do the same for the size of the any text that is displayed on the screen so that it fits!

DustinWatts, 2021-01-02:
I resolved this in https://github.com/DustinWatts/FreeTouchDeck/commit/8b1e782ab99d6dabfde170a4ac81ebcca9947a5d If this works for you, you may close the issue ;)

charliejgallo, 2021-01-02:
0.9.7 works great!!!! (just had to update the esp32-ble-keyboard for the version number update)

closed 2021-01-02 by charliejgallo