DustinWatts/FreeTouchDeck #23: WiFi Config Failed to load an BT-Connetion is lost in 20sec.

bikejoco, 2021-01-01:
Hello!
Thanks for your hard work to create this usefull Project.

I used this Instruction: https://www.instructables.com/A-Bluetooth-ESP32-TFT-Touch-Macro-Keypad/

But I can't start the WiFi Config. The wificonfig.json is correct modified with my SSID and PW. The TouchDeck initializied well and ich can touche the Buttons. In the BT-Menue on my Mac (10.12.6) I can see and connect to the FreeTouchDeck but the connction is lost in 20sec. In the Keyboard-Setting I find an can connect to the TouchDeck but this connection also lost quick.

Do you have an idea for me to solve this Problems?

Thanks for Help and best greatings from Germany
JOHANNES

https://user-images.githubusercontent.com/65533701/103442515-274aa080-4c57-11eb-8d87-c0519ce7813c.mp4

DustinWatts, 2021-01-01:
Hi Johannes! It seems the BT connection isn't made at all. Can you also tell what ESP32 Development board you have? 

When the "WiFi Config Failed to load!" message is displayed on the screens it means that for some reason reading the JSON file went wrong. Can you open the serial monitor through the Arduino IDE and post the log from the serial monitor? 

If the JSON file is not in the correct format it can also fail to load. Can you post the wificonfig.json? Off course you can leave the SSID and password out. It is just so I can see if it properly formatted. 

Both BT and loading the JSON file failing are making me think of a memory issue. I can be totally of base here, but let's investigate!

DustinWatts, 2021-01-01:
I just though of something else. I have updated the https://github.com/DustinWatts/ESP32-BLE-Keyboard. Can you install the new version of this library? If it is a memory issue, this will likely fix it.

bikejoco, 2021-01-02:
Thanks for your fast reply. I use this ESP32 Dev Board (aliexpress.com/item/32959541446.html). I tested it on my iPhone and was able to controle volume und mute. There ist works. Now I installed your update. Now it is permanet connected to my MacBooc via BT (seen in taskbar) as an keyboard. But the mute and volume function doesn't work. I also tried to fix it with a new upload of the "ESP32 Sketch Data upload" (the "DS_Store" was skipped???):

[SPIFFS] data   : /Users/Johannes/Documents/Arduino/FreeTouchDeck/data
[SPIFFS] start  : 2162688
[SPIFFS] size   : 1984
[SPIFFS] page   : 256
[SPIFFS] block  : 4096
/config/colors.json
/config/homescreen.json
/config/menu1.json
/config/menu2.json
/config/menu3.json
/config/menu4.json
/config/menu5.json
/config/wificonfig.json
/editor.htm
/error.htm
/favicon.ico
/freetouchdeck_logo.bmp
/index.htm
/jquery-3.5.1.slim.min.js
/logos/brightnessdown.bmp
/logos/brightnessup.bmp
/logos/dslr.bmp
/logos/firefox.bmp
/logos/home.bmp
/logos/info.bmp
/logos/mail.bmp
/logos/micmute.bmp
/logos/music.bmp
/logos/mute.bmp
/logos/obs.bmp
/logos/play.bmp
/logos/question.bmp
/logos/settings.bmp
/logos/sleep.bmp
/logos/spanner.bmp
/logos/splash.bmp
/logos/stop.bmp
/logos/volumedown.bmp
/logos/volumeup.bmp
/logos/webcam.bmp
/logos/wifi.bmp
/logos/youtube.bmp
/saveconfig.htm
/upload.htm
**skipping .DS_Store**
[SPIFFS] upload : /var/folders/ss/9l9c7m692952yjwhsgc2mz000000gn/T/arduino_build_815942/FreeTouchDeck.spiffs.bin
[SPIFFS] address: 2162688
[SPIFFS] port   : /dev/cu.SLAB_USBtoUART
[SPIFFS] speed  : 921600
[SPIFFS] mode   : dio
[SPIFFS] freq   : 80m

esptool.py v2.6
Serial port /dev/cu.SLAB_USBtoUART
Connecting.......
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: ac:67:b2:36:4f:c4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 2031616 bytes to 152346...
Wrote 2031616 bytes (152346 compressed) at 0x00210000 in 11.7 seconds (effective 1392.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

this is my wificonfig.json:
[image]

This is the Log:
[INFO]: Loading version 0.9.6
[INFO]: All configs loaded
[INFO]: General logo's loaded.
[INFO]: Starting BLE
[INFO]: Drawing keypad

Thanks for your help
JOHANNES

DustinWatts, 2021-01-02:
Thanks for posting your wificonfig.json. The problem in that file is that the `„` should be regular double quotes `"`. That should take care of the WiFi issue. 

As to the BT issue. I have a feeling that it has something to do with your MacBook / MacOS version as you say on your phone it is working. A few things to check are:

- Does your MacBook support BLE? (Go to the **"About this Mac"**, click on **"System Report"**,  the **"Bluetooth"** under hardware. If you click on it you can get some information about the Bluetooth module. There should be a line saying "Bluetooth Low Energy Supported : Yes".
- If you remove the device, stop and start Bluetooth, does the problem still exist?
- Do you have access to another computer (Mac/Windows/Linux) and does it work there?
- If you reset the Bluetooth module on your Mac does the problem still exist? (A guide on how to reset the BT on your Mac can be found here: https://www.macrumors.com/how-to/reset-mac-bluetooth-module/)

closed 2021-10-15 by DustinWatts