DustinWatts/FreeTouchDeck #26: extended audio functionality

gufpewga, 2021-01-15:
Hi,
I really like this project. And wa thinking if it was posible to add some funcionality like on GOXLR to control audio. I mean that if you for example swipe left on main screen it takes you to diferent layout that will have sliders with different audio sources. It is just idea, don't know if it is doable. But would be cool to have 2 in 1 that doesnt take too much space on desk.

DustinWatts, 2021-01-15:
Hi @gufpewga ! Unfortunately not all touch controller support gestures natively. That would mean some sort of software "hack" to make that possible. For now this is beyond the scope of FreeTouchDeck. But I agree that this would be a great feature!

briight, 2021-02-13:
im already have a physical volume controller and im looking at intregrating it into/on my freetouchdeck   just waiting for parts 

this is the one i am using atm https://github.com/omriharel/deej

briight, 2021-02-13:
i have posted some pics on the discord  look up briight

closed 2021-10-15 by DustinWatts