DustinWatts/FreeTouchDeck #28: White screen on boot

souldessin, 2021-01-29:
Hello, I would first like to thank you for making this project. I ordered all of the recommended items, including the board from PCBway.

My issue is with the FreeTouchDeck showing a white screen on boot. I followed everything and tried double checking my setup for the Arduino IDE on Windows 10. There are no errors on compiling and the only sign that some of the sketch has uploaded, is that the screen will go dark after 10 minutes. Other than that, there is no response from the unit and there is no available bluetooth device on the desktop, or my phone, after boot.

I'm not the best with microcontrollers, and I'm at a bit of a loss. Is there any recommendation that you may have for resolving this?

DustinWatts, 2021-01-30:
Hi @souldessin Can you post the user_setup.h file from the TFT_eSPI library?

That is where the pins are defined that are used by the screen. Maybe that is where it has gone wrong?

souldessin, 2021-01-31:
Alright, I retired everything on a linux computer AND this time I used the example User_Setup.h file. Everything worked perfectly. 

I think that your assumption was correct with the file. 

Thanks again for the great project.

closed 2021-01-31 by souldessin

DustinWatts, 2021-01-31:
@souldessin No problem! Glad it worked out in the end. Have fun!