DustinWatts/FreeTouchDeck #30: Battery powered???

gufpewga, 2021-02-09:
Hi,
I really enjoy this project, but would like to use it with battery power.
I saw you wrote that you can monitor battery level in this issue #8.
Is it working, and which pin is used for battery monitoring?
And not that important, but will it show percentage in Windows?

DustinWatts, 2021-02-09:
Hi @gufpewga Battery monitoring is not supported out of the box. But there are a few people who have made battery powered FTD's with external charge modules. There are enough analog inputs available so you could use any the pins not used by FreeTouchDeck and use a voltage divider to sense the voltage. 

So yes, it can definitely be battery powered. The blekeyboad library has a option for sending battery levels to the host computer but I do not know if that works on Windows.

Rowdydangerous, 2021-02-18:
@gufpewga is this a closed issue now?

DustinWatts, 2021-02-18:
No not quite. Battery powerable = definitely yes! Sending battery level through bleKeyboard = untested.

gufpewga, 2021-02-18:
Srry for my late reply. I have it kinda working. It works sometime and sometime dont :D 
I used 18650CL libary for checking battery level and then it gets updated if it changes.
[image]

DustinWatts, 2021-02-18:
@gufpewga That is great! Can you share the code who you did it? If this can be tweaked to work most of the time, I'd like to add it to a future release!