DustinWatts/FreeTouchDeck #35: Question: Fusion360 case

PeterEIER, 2021-02-21:
Hi Dustin, it seems to me that the Fusion360 design is missing the back/stand of FreeTouchDeck case. When I import the f3z file in to Fusion360 I get 2 drawing, the front of the case and dummy ILI9488 TFT screen.

My question is this... would it be possible to also share the back/stand?

Thx.

DustinWatts, 2021-02-22:
Hi Peter! The FreeTouchDeck Fusion360 file contains both the front and the back, each is in it's own component.
[image]

You can show/hide the ones you want by using the little eye to the left of each component.

PeterEIER, 2021-02-22:
ouch, my wrong... you are completely right... sorry for bothering you!

FYI - I will make some changes to the original case to have 2 FreeTouchDecks in just 1 case ;-)

Thx, great project!

DustinWatts, 2021-02-22:
@PeterEIER No problem, you are welcome! Looking forward to see your MultiTouchDeck! ;)

closed 2021-02-22 by DustinWatts