DustinWatts/FreeTouchDeck #37: obs out of focus commands

briight, 2021-03-04:
just did some testing and looks like  keypresses must have a 100ms delay added after your key for obs to pick it up 

maybe something put into the readme about this 

or maybe adding 100ms delay to FTD wide  but i only see this as an option if its only obs that has the problem
[image]

DustinWatts, 2021-03-04:
Great to hear that that works for you @briight. I though I already tried this and it didn't work for me. If this is the solution I'm happy that no other app is needed to get OBS in to focus!. I'll keep this issue open so other people who struggle with this can find it quicker!

briight, 2021-03-06:
also found that if i have a hotkey of    f13     and i try a combo   like  shift + f13    it will trigger both hotkeys,
so to get more hotkeys  use  alt + f13 , shift + f13 , ctrl + f13  and they will all work independly of eachother 
[image]

kicker22004, 2021-05-01:
Found that I couldn't set a scene hotkey to (Shift+/) and it turns out I couldn't detect any shift to OBS. I'm thinking because OBS doesn't show left or right.. Ideas?

DustinWatts, 2021-05-05:
@kicker22004 This is something I have not experienced but I can see if I can try and replicate that. Can you tell me if your computer does catch that macro? With some thing like a keyboard viewer? `SHIFT+/` is actually `?`so maybe OBS registers it as 'SHIFT+/` but bleKeyboard is actually sending a `?`. I don't know, just a wild guess...