DustinWatts/FreeTouchDeck #38: Missing keys

rienboonstoppel, 2021-03-14:
Hi, I really enjoy the touchdeck.
However, I am missing some keys to map to in the configurator.
It seems like the , (comma), . (point) and the - (minus) is missing. Is it possible to add those? 
For now, I took the long route via a AHK script and extended Function keys, but it would be great to have those inside the configurator. 
I use it to improve my Lightroom workflow :)

Thanks in advance.

DustinWatts, 2021-03-14:
Hi @rienboonstoppel Absolutely! I've marked this issue so I don't forget! :)

briight, 2021-03-15:
also a fkey missing and one miss labled

DustinWatts, 2021-03-15:
@briight and @rienboonstoppel both are fixed in version 0.9.11 https://github.com/DustinWatts/FreeTouchDeck/commit/0f8782ed4176e20e47ef9934b29f748d2912b787 

@rienboonstoppel If you are happy, you may close this issue :)

closed 2021-03-15 by rienboonstoppel