DustinWatts/FreeTouchDeck #40: Set the BLE name through configurator

DustinWatts, 2021-03-17:
The idea is that if you have multiple FreeTouchDecks you can set a BLE name in the configurator, to keep them apart.

For now the only possibility for setting the device name for BLE is during compiling.  It would require some changes to the BLE Keyboard library, but I think this is possible.

jptrsn, 2021-03-24:
The default could also include the chip ID as an identifier. For example, `FreeTouchDeck-A74DE8`

DustinWatts, 2021-03-24:
@jptrsn That is a good idea!

danielhunt, 2021-05-18:
> For now the only possibility for setting the device name for BLE is during compiling. It would require some changes to the BLE Keyboard library, but I think this is possible.

It looks like the device name can be updated before `bleKeyboard.begin()` is called [at setup](https://github.com/DustinWatts/FreeTouchDeck/blob/master/FreeTouchDeck.ino#L545)
```
bleKeyboard.deviceName = "SuperFreeTouchDeck";
bleKeyboard.begin();
```

gjelsoe, 2021-05-31:
@danielhunt It's possible as you wrote, did it in my "Fork" and it's configurable in the Configurator.

sle118, 2021-06-03:
@jptrsn I've implemented something similar in my squeezelite-esp32 project. I'll just go ahead and steal my own code to implement in my own fork.

Still very much in Alpha stage, but worth a try if you are interested
https://github.com/sle118/FreeTouchDeck

closed 2022-06-07 by github-actions[bot]