DustinWatts/FreeTouchDeck #41: ILI9488 - MSP3520 with Touch

schneika, 2021-03-21:
I have ordered the TFT with Touch from AliExpress. Version MSP3520. The TFT Display shows the Icon, but the Touch is not working. I dont no what Touch-Chip is the MSP3520 Version...

DustinWatts, 2021-03-21:
@schneika Is the chip the XPT2046? The screen also come without touch, so you have to make sure they have the film on top with a silver boarder. Also U2 (the chip) should be populated. 

The first time FTD boots, it does a calibration. If that fails or does not calibrate o.k. and touch doesn't work, you can open the serial monitor (assuming you used the Arduino IDE), set the baud rate to 9600, and make sure you've selected "No line ending" and type `cal reset`, to calibrate again.

schneika, 2021-03-21:
Yes, the Chip is the XPT2046. 
And thanks for your help.: After recalibration as suggested from you (via Serial Monitor - cal reset) - the Touch-Controller is working as expected. Fantastic work!

DustinWatts, 2021-03-21:
@schneika Good to hear! :D

closed 2021-03-21 by DustinWatts