DustinWatts/FreeTouchDeck #42: ConfigLoad.h:20:3: error: 'DynamicJsonDocument' was not declared in this scope

gadjet, 2021-03-24:
Hi,
I followed the instructions in the vide, installed all the libraries but when I compile I get the error
"ConfigLoad.h:20:3: error: 'DynamicJsonDocument' was not declared in this scope"
I'm using Arduino IDE 1.8.42.0
I have ArduinoJson 6.17.3 installed.
Anybody know what may be causing this, I do have quite a few erros after this but not sure if they are a result of this error.

Thanks

DustinWatts, 2021-03-25:
Hi @gadjet Can you try and compile the `StringExample` example of the ArduinoJson library? This is a good indicator whether the library is correctly installed for your board. Also which board did you compile for?

gadjet, 2021-03-25:
HI, String example compiles OK and I have "ESP32-WROOM-32D" from Aliexpress and selected "ESP32 Dev Kit" in the board manager.

>

DustinWatts, 2021-03-25:
I just tried it (ArduinoJson 6.17.3 and ArduinoIDE 1.8.13 (MacOS)) and it compiles fine for me. Did you already try and re-download the latest version of FreeTouchDeck? Could you post the complete output log, maybe there is a clue in there?

gadjet, 2021-03-25:
Thanks, I'll give it a go later and let you know what happens.

gadjet, 2021-03-29:
Right I think I found the problem, there wwere two installed libraries for ArduinoJson and it was trying to use an earlier one, I deleted the older library and it has compiled.

Thanks for the help

closed 2021-03-29 by gadjet