DustinWatts/FreeTouchDeck #43: 'File' was not declared in this scope

seamus65, 2021-03-25:
I am attempting to compile version number 0.9.11 and i am getting the errors:

ScreenHelper.h:244:23: error: 'File' was not declared in this scope

ScreenHelper.h:244:29: error: 'p_file' was not declared in this scope

ScreenHelper.h:244:37: error: expected primary-expression before 'int'

ScreenHelper.h:244:56: error: expected primary-expression before 'nBytes'

ScreenHelper.h:244:62: error: expression list treated as compound expression in initializer [-fpermissive]

ScreenHelper.h:245:1: error: expected ',' or ';' before '{' token

I am not sure if I am doing something wrong?

seamus65, 2021-03-25:
I think I had doubled up on include libraries, sorry.

closed 2021-03-25 by seamus65

DustinWatts, 2021-03-25:
No problem :)

2Gapps, 2021-03-27:
Hello @seamus65,

I have a similar error message. Can you please tell me how did you solve this error?

DustinWatts, 2021-03-27:
@2Gapps If you post the exact error messages you are getting that might help. Or just insert the Arduino log here using "Insert code" ( looks like this < >).

seamus65, 2021-03-27:
Hello @2Gapps,

I had included the libraries a second time by accident, so I removed the duplicates and that removed the error.
It is also worth checking you have the correct lines uncommented in the TFT_eSPI library user_setup.h file and comment out the lines not required.

2Gapps, 2021-03-27:
Thank you @seamus65 and  @DustinWatts  for your quick response.  This is the error message that I am getting. 

--------------------------------------
--- Error Message 
--------------------------------------
Arduino: 1.8.13 (Windows 10), Board: "ESP32 Dev Module, Disabled, No OTA (2MB APP/2MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from C:\Program Files (x86)\Arduino\FreeTouchDeck\FreeTouchDeck.ino:297:0:

ScreenHelper.h:244:23: error: 'File' was not declared in this scope

 int32_t readNbytesInt(File *p_file, int position, byte nBytes){

                       ^

sketch\ScreenHelper.h:244:23: note: suggested alternative:

In file included from D:\Documentos\Arduino\libraries\TFT_eSPI-master/Processors/TFT_eSPI_ESP32.h:78:0,

                 from D:\Documentos\Arduino\libraries\TFT_eSPI-master/TFT_eSPI.h:52,

                 from C:\Program Files (x86)\Arduino\FreeTouchDeck\FreeTouchDeck.ino:1:

C:\Users\DGAMBOA\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FS\src/FS.h:47:7: note:   'fs::File'

 class File : public Stream

       ^

In file included from C:\Program Files (x86)\Arduino\FreeTouchDeck\FreeTouchDeck.ino:297:0:

ScreenHelper.h:244:29: error: 'p_file' was not declared in this scope

 int32_t readNbytesInt(File *p_file, int position, byte nBytes){

                             ^

ScreenHelper.h:244:37: error: expected primary-expression before 'int'

 int32_t readNbytesInt(File *p_file, int position, byte nBytes){

                                     ^

ScreenHelper.h:244:56: error: expected primary-expression before 'nBytes'

 int32_t readNbytesInt(File *p_file, int position, byte nBytes){

                                                        ^

ScreenHelper.h:244:62: error: expression list treated as compound expression in initializer [-fpermissive]

 int32_t readNbytesInt(File *p_file, int position, byte nBytes){

                                                              ^

ScreenHelper.h:244:63: error: expected ',' or ';' before '{' token

 int32_t readNbytesInt(File *p_file, int position, byte nBytes){

                                                               ^

exit status 1

'File' was not declared in this scope

--------------------------------------
--- libraries
--------------------------------------
D:\Documentos\Arduino\libraries
Adafruit_BusIO
Adafruit_Circuit_Playground
Adafruit_GFX_Library
Adafruit_GPS_Library
Adafruit_ILI9341
Adafruit_LED_Backpack_Library
Adafruit_SleepyDog_Library
Adafruit_STMPE610
Adafruit_TouchScreen
Adafruit_Zero_DMA_Library
Adafruit_Zero_FFT_Library
Adafruit_Zero_PDM_Library
ArduinoJson
AsyncTCP-master
ESP32-BLE-Keyboard-master
ESPAsyncWebServer-master
Generar_Lista_new.cmd
readme.txt
RTClib
Servo
TFT_eSPI-master
TinyWireM
WaveHC

---
C:\Program Files (x86)\Arduino\libraries
Adafruit_Circuit_Playground
Bridge
Esplora
Ethernet
Firmata
GSM
Keyboard
LiquidCrystal
Mouse
Robot_Control
Robot_Motor
RobotIRremote
SD
Servo
SpacebrewYun
Stepper
Temboo
TFT
-----
User_Setup.h <- ESP32_Dev_Kit_V1_ILI9488_Resistive.h

Thank you in advance.

2Gapps, 2021-03-27:
I have found the error.  Possibly an error on my part when reviewing the content of the .h documents.

I read this article https://forum.arduino.cc/index.php?topic=526770.0, unzip the project again and everything works correctly.

Thanks.

DustinWatts, 2021-03-27:
@2Gapps Good to hear! :)