DustinWatts/FreeTouchDeck #45: Missing keys (If you find them, report them here)

DustinWatts, 2021-03-28:
A list so far:

- ESC
- PrtScn
- Home
- End

TheAlgorythm, 2021-05-19:
I would need ESC and custom text to interact with vim.

DustinWatts, 2021-05-19:
> I would need ESC and custom text to interact with vim.

Can you explain a bit more about the custom text?

TheAlgorythm, 2021-05-19:
Well, would it be possible to print some characters (like the length of a word) out. e.g. ":Buffer" or any other command. Another use-case for me (and maybe for others also?) would be to type common or hard to reach code out (like `\begin` in `LaTeX`).

DustinWatts, 2021-05-20:
Yes that is possible. But it is not possible in the configurator. You would have to hard code that in FreeTouchDeck.ino. This is a future feature because indeed other people also would like this.

DustinWatts, 2021-05-20:
https://github.com/DustinWatts/FreeTouchDeck/commit/faa9d31d738b3f14bbc285f9732b96536a5d19ab adds ESC, HOME, END, PrintSreen.

danielo515, 2021-11-11:
Well, in the same spirit as @TheAlgorythm I also would love to be able to configure custom MACROS that are just send as keystrokes.
On top of that, i think that the entire list of characters needs to be included, at least all the characters that can be typed with a keyboard: `[]~@<>{}()=: ; _ ` are just some examples that I use daily

DustinWatts, 2021-11-12:
@danielo515 As I said to TheAlgorythm, it is possible, only not through the configurator. You will have to edit the code and create custom macros. 

I do take notice of the characters and will add them as soon as possible.

danielo515, 2021-11-12:
> @danielo515 As I said to TheAlgorythm, it is possible, only not through the configurator

Well, thanks for confirming it is possible on the code, but one of the things that attracted me to the touch-deck was being able to quickly iterate on macros, as opposed to my QMK macropads that require a compilation step.
So my vote is to find a way to do that on the GUI.

DustinWatts, 2021-11-13:
@danielo515 https://github.com/DustinWatts/FreeTouchDeck/commit/3aa366427ff764bf857e26cfc1cbb0534e1efc70 adds the missing keys.

Also, a new file is there called UserActions.h. In here there are functions that you can create (userAction1(); etc) that will let you put the whole function under a button via the configurator. Have a look at the examples!

danielo515, 2021-11-15:
That looks awesome @DustinWatts !!!
Thank you very much

closed 2022-06-07 by github-actions[bot]