DustinWatts/FreeTouchDeck #46: Freetouchdeck doesn't connect to my WiFi network

gadjet, 2021-03-29:
Having got it working I can go into the WiFi settings and connect direct to the Freetouchdeck with my phone but even after setting my wifi details and password it doesn't seem to want to connect to my WiFi network so that I can access the setup from my main PC.
I do have other IOT devices that are connected and accessible to my wifi though so not sure what's going on.

briight, 2021-03-29:
Ssid the exactly the same.  Caps And all ?

briight, 2021-03-29:
Try setting up a hotspot on your phone and see if u it will connect to that

DustinWatts, 2021-03-30:
@ gadjet What version are you running? The latest version automatically switches to AP mode when it can't connect to an AP. After that you can connect to it directly and increase the number of attempts and delay between attempts to connect to an AP.

gadjet, 2021-03-30:
> Ssid the exactly the same. Caps And all ?

Yep, checked multiple times :-)

gadjet, 2021-03-30:
> Try setting up a hotspot on your phone and see if it will connect to that

Worth a try good idea

gadjet, 2021-03-30:
> @ gadjet What version are you running? The latest version automatically switches to AP mode when it can't connect to an AP. After that you can connect to it directly and increase the number of attempts and delay between attempts to connect to an AP.

It's the latest as far as I can see, I can connect to it directly when it is in AP mode, I'll increase the time between retries and see if it helps.

gadjet, 2021-03-30:
OK, I changed the connection retry from 500ms to 2 seconds and it connects now.

closed 2021-03-30 by gadjet

gadjet, 2021-03-30:
> @gadjet Good to hear! It might be an idea for me to increase the default timeout and delay in the code. I see no reason for it to not be longer by default.

I think that would be a good idea :-)