DustinWatts/FreeTouchDeck #47: Allow main menu to trigger applications

gadjet, 2021-03-31:
I've got it all up and running now, thanks for a great project.

Is it possible for the home menu to trigger some actions as well as going to a sub menu?

My scenario is that I want to start Fusion360 by pressing a button on the home (I've assigned CTRL+ALT+f to the shortcut) then the sub menu for Fusion actions will appear where I will have buttons to do basic CAD actions.

Also it would be useful if we could have more buttons, maybe 4 x 3 ?

DustinWatts, 2021-03-31:
Hi @gadjet That is a good idea. I'll have to think about a good way to implements this in the configurator. But you can do this yourself for now in the code. At this line https://github.com/DustinWatts/FreeTouchDeck/blob/5c46b69fc5a63dbbbeb122133aec389d7aa6c7f0/FreeTouchDeck.ino#L1032
the home menu buttons are handled. You could add some code that sends key presses. 

The bigger matrix is a much requested feature. I am quite busy atm with some other things and it will take quite a while to make the matrix scalable, but it is high on my priority list!

gadjet, 2021-03-31:
Great, I'll give that a try in the code.
No hurry for the other request :-)

gadjet, 2021-03-31:
@DustinWatts Your work around works a treat, thanks, noe to make some button logos :-)

GabinTB, 2021-05-23:
Hi @gadjet, can you give details about what you did there to open App from the FreeTouchDeck ?

gadjet, 2021-05-23:
> Hi @gadjet, can you give details about what you did there to open App from the FreeTouchDeck ?

No problem, right click on an application icon on your desktop and click properties, in the properties there is a shortcut TAB and on that TAB there is a section for "Shortcut key", click in the box and then use the shortcut you want to assign to it then add this in the code as described by DustinWatts above.