DustinWatts/FreeTouchDeck #50: flash read err, 1000

seamus65, 2021-04-08:
I purchased the ESP32 + TFT Combiner V1 from PCBWay and soldered the ESP32 and the TFT on, but now I am getting the below error through serial and the screen never boots up.

I tested using a breadboard and jumper wires before soldering to the PCB and everything worked great.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

DustinWatts, 2021-04-08:
Hi @seamus65, a few things you can do:

- Try a different power supply
- Make sure there are no solder shorts on the PCB. 
- Do you have any other things connected to the ESP32 pins? Try it without.

Since the error did not show up after you soldered the ESP32 and TFT to the board, my guess is on a solder bridge. Pay special attention to pin 0,2,5, and 12. Just to be sure you may want to reflow those.

seamus65, 2021-04-08:
Is it normal to not be able to flash with the ESP32 attached to the PCB?

I tried a second ESP32 and inserted it into another PCB, not soldered only inserted, and it wouldn't flash, but when I removed it from the PCB it flashed no problem.

DustinWatts, 2021-04-08:
No that is not normal. What ESP32 are you using? I might have happened that there were manufacture errors on the board. Maybe you can join me on Discord so we can find out what is going wrong? https://discord.gg/RE3XevS