DustinWatts/FreeTouchDeck #53: Button 6 Defaults to home.bmp

daedalas1981, 2021-05-12:
Button 6 changes successfully, but upon reloading the menu, or restarting, the json is not being updated with the new icon.

DustinWatts, 2021-05-12:
`home.bmp` is hard coded. There is no reason this should be like it is. A way around this is uploading the home icon you want as home.bmp, this will overwrite the original `home.bmp`.

daedalas1981, 2021-05-12:
I had setup macros for a chrome web browser. Changed my icons. But I came across the issue where I was using the home.bmp for the shortcut to the homepage (Also had to make custom hotkeys since there was no Home keymap). Using the webserver, I used the drop down for the Button 6, and used a back arrow to replace the home.bmp for Button 6. That's when I noticed it was hard coded, and would revert back to home.bmp.

pearson, 2021-07-17:
If a change to button 6 does not save, perhaps the configuration page shouldn't allow the user to change it? That confused me as well, and led me to come here to look for a new version of the software.

lbehrler, 2021-08-30:
It would be nice to code this button to a back arrow or in my case I would like to code it to a custom home button.  I have 5 setups a couple of which are for meetings and such It would be nice to use that 6th position with a custom icon to give the user and idea what sub-menu they are in.

closed 2022-06-06 by github-actions[bot]