DustinWatts/FreeTouchDeck #54: General.json

daedalas1981, 2021-05-12:
If you use the Upload Menu Config to upload general.json, you get an error that you can't upload it.

DustinWatts, 2021-05-12:
Ah, I see! This is because checks if the filename that is uploaded is one that is used in the config. Since 0.9.11 the config filesnames have changed a bit but I forgot to change the allowed filenames in `Webserver.h`. The same would happen if you try to upload wificonfig.json. Quick fix:

`line 223 Webserver.h` should be:
```
if (filename != "menu1.json" && filename != "menu2.json" && filename != "menu3.json" && filename != "menu4.json" && filename != "menu5.json" && filename != "homescreen.json" && filename != "general.json", && filename != "wificonfig.json")
 ```

daedalas1981, 2021-05-12:
Awesome, thanks! Updated code, will try later!

DustinWatts, 2021-05-20:
https://github.com/DustinWatts/FreeTouchDeck/commit/faa9d31d738b3f14bbc285f9732b96536a5d19ab Fixes this.

closed 2021-05-20 by DustinWatts