DustinWatts/FreeTouchDeck #55: Touch screen malfunctioning cal reset does not work.

Lucero-delcielo, 2021-05-12:
I have been using my FTD for a few days now, and it has been working great, but today I tried to use it and the touch screen would work a little but seemed to be making its own touches. I ran cal reset a couple times, but it seemed to just touch each corner on its own without me touching it at all. I am at a loss. Thoughts?

DustinWatts, 2021-05-12:
Hi @Lucero-delcielo, my first guess is a bad connection to the touch controller. Are you using a breadboard? Or the combiner PCB? Or a prototype board? Anyway, I would check the T_DO, T_CS, and T_CLK for a bad connection or a short. Maybe even reflow the pins if you are using the combiner board or a prototype board.

Lucero-delcielo, 2021-05-26:
Thank you for the advice Dustin. It seems my soldering was a bit shoddy on those pins. Reflowing them solved the issue. Thanks so much for your help!

DustinWatts, 2021-05-26:
Good to hear! :D

closed 2021-05-26 by DustinWatts