DustinWatts/FreeTouchDeck #59: Touchscreen calibration not working

danielhunt, 2021-05-18:
@DustinWatts do you happen to have the values for the calibration array handy?
While uploading new contents for the `data` directory (while iterating on #57) I seem to have accidentally wiped the existing config 😢 

If you have them handy I can just overwrite the existing values 👍

danielhunt, 2021-05-18:
I forgot to add: While rebooting the device prompts for me to calibrate the screen, but it doesn't recognise, respond to, or store any boundary information, so touch doesn't work after boot is complete

DustinWatts, 2021-05-18:
I do not have that data ready for you as that is completely done by code and always lives in the SPIFFS. But you can re-calibrate whenever you want by typing `cal reset` (technically cal<space> is enough) in the serial monitor (No Line Ending). This will erase the calibration file, reboot, and start the calibration over again. 

Just realized, are you using the ESP32 TouchDown? If so, you do not need calibration as this is the capacitive touch screen. If you uncomment `#define USECAPTOUCH` works without calibration (just make sure you have the FT6236 library installed).

DustinWatts, 2021-05-18:
No problem, I though I remembered you were using the ESP32 TouchDown :)