DustinWatts/FreeTouchDeck #62: Compiling error...

neolinux, 2021-06-15:
Hello,

i have these errors when compiling the code in arduino:

`In file included from C:\Users\neolinux\Documents\Arduino\libraries\TFT_eSPI/User_Setup_Select.h:22:0,
                 from C:\Users\neolinux\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:39,
                 from C:\Users\neolinux\Downloads\Internet_Download_Manager\Compressed\FreeTouchDeck\FreeTouchDeck.ino:79:
C:\Users\neolinux\Documents\Arduino\libraries\TFT_eSPI/User_Setup.h:199:0: warning: "TFT_CS" redefined
 #define TFT_CS   15  // Chip select control pin
 ^
C:\Users\neolinux\Documents\Arduino\libraries\TFT_eSPI/User_Setup.h:161:0: note: this is the location of the previous definition
 #define TFT_CS   PIN_D8  // Chip select control pin D8
 ^
C:\Users\neolinux\Documents\Arduino\libraries\TFT_eSPI/User_Setup.h:200:2: error: stray '#' in program
 /#define TFT_DC    2  // Data Command control pin
  ^
C:\Users\neolinux\Documents\Arduino\libraries\TFT_eSPI/User_Setup.h:201:2: error: stray '#' in program
 /#define TFT_RST   4  // Reset pin (could connect to RST pin)
  ^
In file included from C:\Users\neolinux\Documents\Arduino\libraries\TFT_eSPI/User_Setup_Select.h:22:0,
                 from C:\Users\neolinux\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:39,
                 from C:\Users\neolinux\Downloads\Internet_Download_Manager\Compressed\FreeTouchDeck\FreeTouchDeck.ino:79:
C:\Users\neolinux\Documents\Arduino\libraries\TFT_eSPI/User_Setup.h:200:1: error: expected unqualified-id before '/' token
 /#define TFT_DC    2  // Data Command control pin
 ^
C:\Users\neolinux\Downloads\Internet_Download_Manager\Compressed\FreeTouchDeck\FreeTouchDeck.ino:268:22: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
 char* jsonfilefail = "";
                      ^
In file included from C:\Users\neolinux\Downloads\Internet_Download_Manager\Compressed\FreeTouchDeck\FreeTouchDeck.ino:283:0:
sketch\DrawHelper.h: In function 'void drawKeypad()':
sketch\DrawHelper.h:1345:43: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
                           "", KEY_TEXTSIZE);
                                           ^
sketch\DrawHelper.h:1402:45: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
                             "", KEY_TEXTSIZE);
                                             ^
sketch\DrawHelper.h:1463:45: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
                             "", KEY_TEXTSIZE);
                                             ^
C:\Users\neolinux\Downloads\Internet_Download_Manager\Compressed\FreeTouchDeck\FreeTouchDeck.ino: In function 'void setup()':
C:\Users\neolinux\Downloads\Internet_Download_Manager\Compressed\FreeTouchDeck\FreeTouchDeck.ino:450:18: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     jsonfilefail = "general";
                  ^
C:\Users\neolinux\Downloads\Internet_Download_Manager\Compressed\FreeTouchDeck\FreeTouchDeck.ino:484:18: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     jsonfilefail = "homescreen";
                  ^
C:\Users\neolinux\Downloads\Internet_Download_Manager\Compressed\FreeTouchDeck\FreeTouchDeck.ino:490:18: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     jsonfilefail = "menu1";
                  ^
C:\Users\neolinux\Downloads\Internet_Download_Manager\Compressed\FreeTouchDeck\FreeTouchDeck.ino:496:18: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     jsonfilefail = "menu2";
                  ^
C:\Users\neolinux\Downloads\Internet_Download_Manager\Compressed\FreeTouchDeck\FreeTouchDeck.ino:502:18: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     jsonfilefail = "menu3";
                  ^
C:\Users\neolinux\Downloads\Internet_Download_Manager\Compressed\FreeTouchDeck\FreeTouchDeck.ino:508:18: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     jsonfilefail = "menu4";
                  ^
C:\Users\neolinux\Downloads\Internet_Download_Manager\Compressed\FreeTouchDeck\FreeTouchDeck.ino:514:18: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     jsonfilefail = "menu5";
                  ^
C:\Users\neolinux\Downloads\Internet_Download_Manager\Compressed\FreeTouchDeck\FreeTouchDeck.ino: In function 'void loop()':
C:\Users\neolinux\Downloads\Internet_Download_Manager\Compressed\FreeTouchDeck\FreeTouchDeck.ino:948:43: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
                           "", KEY_TEXTSIZE);
                                           ^
C:\Users\neolinux\Downloads\Internet_Download_Manager\Compressed\FreeTouchDeck\FreeTouchDeck.ino:1013:43: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
                           "", KEY_TEXTSIZE);
                                           ^
exit status 1
Errore durante la compilazione per la scheda ESP32 Dev Module.
`

DustinWatts, 2021-06-15:
It looks like you forgot to comment out some lines in `User_Setup.h`:

```
#define TFT_CS   PIN_D8  // Chip select control pin D8
#define TFT_DC   PIN_D3  // Data Command control pin
#define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)
```

The easiest way to get a working `User_Setup.h` is to copy the contents of the example setup:
```https://github.com/DustinWatts/FreeTouchDeck/blob/master/user_setup.h%20Examples/ESP32_Dev_Kit_V1_ILI9488_Resistive.h```

neolinux, 2021-06-15:
Ok, i've done that, but what abount the warnings:

warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings] "", KEY_TEXTSIZE); ^

My screen is white and nothing is displayed :/

DustinWatts, 2021-06-15:
Those warnings should not affect the functioning of FreeTouchDeck. Are you using the ESP32 TouchDown or the ESP32 in combination a resistive TFT. 

If nothing is displayed it would be worth checking the connections you made.

neolinux, 2021-06-15:
I copied the content of `https://github.com/DustinWatts/FreeTouchDeck/blob/master/user_setup.h%20Examples/ESP32_Dev_Kit_V1_ILI9488_Resistive.h` in the `User_Setup.h`.

I use your PCB for connecting the whole thing.
My Hardware is:

`https://de.aliexpress.com/item/32954128438.html?af=1016336&utm_campaign=1016336&aff_platform=portals-tool&utm_medium=cpa&dp=f74333fe66a4f5bb9f36355a8fd8e9e8&aff_fcid=09613a3b53854c709f90a654b0801489-1623788880805-08563-_ePNSNV&spm=a2g0s.9042311.0.0.17654c4diMz7TD&cv=47843&aff_fsk=_ePNSNV&pvid=288a2217-6824-44d4-b877-801e33d98df9&mall_affr=pr3&sk=_ePNSNV&aff_trace_key=09613a3b53854c709f90a654b0801489-1623788880805-08563-_ePNSNV&rmsg=do_not_replacement&scm=1007.23534.124000.0&terminal_id=07f0b9c1fa714293b9740a793e0dff08&utm_content=47843&utm_source=admitad`

`https://de.aliexpress.com/item/4000296658456.html?spm=a2g0s.9042311.0.0.17654c4diMz7TD`

DustinWatts, 2021-06-15:
are you getting any output on the serial monitor?

neolinux, 2021-06-15:
o_O

it works now, i don't know why... but the touch doesen't work now 🗡️ 

that's what is see in the console:

`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

[INFO]: SPIFFS initialised.
[INFO]: Free Space: 753251
[INFO]: Loading Wifi Config
[INFO]: WiFi Credentials Loaded
[INFO]: Loading version 0.9.13
[INFO]: All configs loaded
[INFO]: General logos loaded.
[INFO]: Starting BLE
[INFO]: BLE Keyboard version: 1.2
[INFO]: ArduinoJson version: 6.18.0
[INFO]: TFT_eSPI version: 2.3.70
[INFO]: Drawing keypad
[INFO]: Sleep enabled.
[INFO]: Sleep timer = 10 minutes
`

DustinWatts, 2021-06-15:
If the touch doesn't work, it probably calibrated wrong. You can type `cal reset` in the serial monitor (No Line Ending) to restart the ESP32 and present the calibration screen again. Uploading the SPIFFS again would also do this...

neolinux, 2021-06-15:
Now it works :D 

Thank You.

DustinWatts, 2021-06-16:
Great!

closed 2021-06-16 by DustinWatts