DustinWatts/FreeTouchDeck #67: Development branch

eduportella, 2021-08-01:
Hey guys!

I was using the master branch here for my setup , but as i saw the video about the new features on development branch i compiled and tried to use, im having a issue here that the menu loads , shows but i cant use any button they have no action , and nothing on the log, if i use the master branch works perfectly , could you please help?

thanks!

eduportella, 2021-08-06:
i tried to upload all again and got the same problem

DustinWatts, 2021-08-06:
Can you explain what you mean with no action? What board are you using?

eduportella, 2021-08-06:
hey @DustinWatts  first of all thanks for answering!

Im using a  ESP32 DEVKIT V1 (WROOM32) and a  TFT LCD module with touchpanel ILI9488 as recomended ([this one with touchscreen](https://a.aliexpress.com/_mrwfep1)).

No Action i was meaning that when i power it up  it shows the bootscreen, loads the main menu but none of the buttons do anything, also no calibration was showed, if i clear all and upload the master version works ok, my user_setup.h is this:

```
/* This is a stripped down version of a User_Setup.h file for TFT_eSPI

Copy the contents of this file to TFT_eSPI-master/User_setup.h 

This version is for a 38-pin ESP32 DevKitC and an ILI9488 screen with resistive touch controller.

*/

#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)

#define TFT_BL   32            // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)

#define TOUCH_CS 21     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.

#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT
#define SPI_FREQUENCY  27000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

```

i have already double checked the connections from the board to the lcd and they are ok

DustinWatts, 2021-08-07:
The development branch has a new UserConfig.h (https://github.com/DustinWatts/FreeTouchDeck/blob/development/UserConfig.h)

If you select the right board there, it should work!

eduportella, 2021-08-07:
hey Dustin, 

I selected the  correct board but still no changes :(
```

#pragma once
#define DEFAULT_LOG_LEVEL LogLevels::DEBUG
//#define PRINT_MEM_INFO
/* ------------------------------------------------------------------------ */
/* Select the board that you are using below. Make sure to only select one! */
/* ------------------------------------------------------------------------ */

//#define MAKERFABTOUCH
//#define ESP32TOUCHDOWN
#define ESP32DEVKIT
//#define ARDUINO_TWATCH

/* ------------------------------------------------------------------------ */
/* Board specific config. No need to touch if you selected the right board  */
/* ------------------------------------------------------------------------ */

/* MakerFab's Touchscreen configuration */
#ifdef MAKERFABTOUCH
#define CUSTOM_TOUCH_SDA 26
#define CUSTOM_TOUCH_SCL 27
#define USECAPTOUCH
#define touchInterruptPin GPIO_NUM_0
#define SCREEN_ROTATION 1
#define INVERSE_Y_TOUCH
#define FLIP_TOUCH_AXIS

/* Arduino TWatch Configuration */
#elif defined(ARDUINO_TWATCH_BASE) && defined(ARDUINO_TWATCH)
#include "axp20x.h"
#define CUSTOM_TOUCH_SDA 23
#define CUSTOM_TOUCH_SCL 32
#define USECAPTOUCH
#define touchInterruptPin GPIO_NUM_38
#define TFT_BL 12
#define SCREEN_WIDTH 240
#define SCREEN_HEIGHT 240
#define SCREEN_ROTATION 2
#define ILI9341_DRIVER

/* ESP32 TouchDown Configuration */
#elif defined(ESP32TOUCHDOWN)
#define touchInterruptPin GPIO_NUM_27
#define speakerPin GPIO_NUM_26
#define SCREEN_ROTATION 1
#define FLIP_TOUCH_AXIS
#define INVERSE_Y_TOUCH
#define USECAPTOUCH
#define SDDAT3 25 

/* Using the ESP32 DevKit with a screen module */
#elif defined(ESP32DEVKIT)
#define SCREEN_ROTATION 1
#define FLIP_TOUCH_AXIS
#define INVERSE_Y_TOUCH
#define touchInterruptPin GPIO_NUM_27
#else 
#error("Unsupported platform")
#endif

/* --- END OF USER CONFIG --- */

#ifndef touchInterruptPin
#define touchInterruptPin -1
#endif
#ifndef speakerPin
#define speakerPin -1
#endif
#if defined(ST7789_DRIVER) || defined(ST7735_DRIVER) || defined(ILI9163_DRIVER)
#define TFT_HEIGHT SCREEN_HEIGHT
#define TFT_WIDTH SCREEN_WIDTH
#endif
#ifndef SCREEN_ROTATION
#define SCREEN_ROTATION 1
#endif
#ifndef INVERSE_Y_TOUCH
#define INVERSE_Y_TOUCH false
#else
#undef INVERSE_Y_TOUCH
#define INVERSE_Y_TOUCH true
#endif

#ifndef INVERSE_X_TOUCH
#define INVERSE_X_TOUCH false
#else
#undef INVERSE_X_TOUCH
#define INVERSE_X_TOUCH true
#endif

#ifndef FLIP_TOUCH_AXIS
#define FLIP_TOUCH_AXIS false
#else
#undef FLIP_TOUCH_AXIS
#define FLIP_TOUCH_AXIS true
#endif

// Text Button Label Font
#define LABEL_FONT &FreeSansBold12pt7b

// Font size multiplier
#define KEY_TEXTSIZE 1

// This is the file name used to store the calibration data
// You can change this to create new calibration files.
// The FILESYSTEM file name must start with "/".
#define CALIBRATION_FILE "/TouchCalData"

// Set REPEAT_CAL to true instead of false to run calibration
// again, otherwise it will only be done once.
// Repeat calibration if you change the screen rotation.
#define REPEAT_CAL false

#define LED_BRIGHTNESS_INCREMENT 25

// percentage of free memory that the system will use as
// a temporary buffer when drawing images. Warning! 
// too much buffer will lead to system instabilities. 
#define BITMAP_BUFFER_FREE_RAM_PCT 0.30
```

eduportella, 2021-08-07:
Oh No :( 

I was trying another board config and my touch display just blinked and now does not work anymore, i checked with a multimeter and the energy is going, i think i will need another one ;(  wait more 40 days

eduportella, 2021-08-10:
I made a. Video already tested the configs as recommended , but still when touch nothing happens neither log

eduportella, 2021-08-11:
Resolved on the third time i deleted all and done it all again started to work! thanks

closed 2021-08-11 by eduportella