DustinWatts/FreeTouchDeck #69: Flashing error ESP32 + TFT

newwolf, 2021-09-10:
Heee,

Saw the new version for flashing by BIN file. (also only the only one now?)
Tried to flash it to my already working TouchDeck, but it gives me a error.
I used multiple ways to flash it, but it keep giving the same error.

Hope you help me/us.

Greetz,
Jos

Using 'COM3' as serial port.
Connecting.....
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 24:0A:C4:EF:80:5C
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)

DustinWatts, 2021-09-11:
Hi @newwolf What are you using for flashing? Can you post the command you are using?

newwolf, 2021-09-11:
I used esphome-flasher and tasmotizer.
The first one uses esptool.py don't know if there is logging

DustinWatts, 2021-09-11:
Try using the esptool.py directly (`esptool.py write_flash 0x0 ESP32DevKitC.bin`). I used that, and got no error.

newwolf, 2021-09-27:
Sorry for the late reaction.
But just tried it and it worked that way. (strange but true)
Thanks.....

C:\temp\esptool-v3.1-win64>esptool write_flash 0x0 ESP32DevKitC.bin
esptool.py v3.1
Found 1 serial ports
Serial port COM3
Connecting......
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:0a:c4:ef:80:5c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x003fffff...
Compressed 4194304 bytes to 1138860...
Wrote 4194304 bytes (1138860 compressed) at 0x00000000 in 101.3 seconds (effective 331.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
C:\temp\esptool-v3.1-win64>

closed 2021-09-27 by newwolf