DustinWatts/FreeTouchDeck #77: Add a checkbox to press all the buttons instead of one after the other

danielo515, 2021-11-11:
Hello.
I just started using my free-touch-deck and I almost immediatly noticed that I can not execute complex combos.
For example `alt + gui + ctrl + ]`. I tried selecting `alt + gui + ctrl` on the first action and then a similar simbol on second action, but they are executed one after the other. I would love to be able to make them be pressed at the same time.

Regards

DustinWatts, 2021-11-12:
Hi @danielo515, if you select `alt+gui+crtl` from the combo's menu, then a symbol on action 2, it should press all of them before releasing. Keys are releases at the end of the three actions. Or if you select release all... I will check if I can replicate your problem...

DustinWatts, 2021-11-12:
> For example `alt + gui + ctrl + ]`. I tried selecting `alt + gui + ctrl` on the first action and then a similar simbol on second action, but they are executed one after the other.

I just confirmed that this configuration presses `alt+gui+crt+]` and then releases them all:
[image]

danielo515, 2021-11-12:
Oh, that is awesome.
Have you added that symbol to the next release already? Wonderful.
Sorry, by reading the docst I thought it was like I described. Glad to see it is not.

closed 2021-11-12 by danielo515

DustinWatts, 2021-11-13:
> Oh, that is awesome. Have you added that symbol to the next release already? Wonderful. Sorry, by reading the docst I thought it was like I described. Glad to see it is not.

No problem!