DustinWatts/FreeTouchDeck #78: Platformio support

danielo515, 2021-11-12:
Hello again, and thanks again for this project.

I was wondering if it would be possible to make this project a little bit more plug-and play than having to manually add a bunch of libraries.
Something like

```
git clone xxxx
cd xxxx
make
make compile
```

will be awesome. I am coming from other modern languages and I am very used to everything living inside each repo, so having to change my local for making one project work is not something I am fan of.
Not sure if that kind of workflow is possible with platformio, but It is the closest thing I know

Regards.

DustinWatts, 2021-11-13:
@danielo515 There are some forks that use PlatformIO (looks like you found one) and it is also on my list :)

DustinWatts, 2021-11-13:
> @danielo515 There are some forks that use PlatformIO (looks like you found one) and it is also on my list :)

Edit: This is a fork you can test https://github.com/pearson/FreeTouchDeck

Before I Pull it though, I want to test is some more. Also, the docs will need some big rewriting. And that is some huge work. Not sure if I will do that before the -development branch becomes the -master branch.

danielo515, 2021-11-15:
> Edit: This is a fork you can test https://github.com/pearson/FreeTouchDeck
>
> Before I Pull it though, I want to test is some more. Also, the docs will need some big rewriting. And that is some huge [...]

I think that is, indeed, the one I found. It needs an update to the docs, and I am not sure about the submodules approach for libraries. I mean, it is better than letting the end-user manage the libraries on his own, but I think using the libraries functionality of platformio will be better

gjelsoe, 2021-12-05:
> Hello again, and thanks again for this project.
>
> I was wondering if it would be possible to make this project a little bit more plug-and play than having to manually add a bunch of libraries. [...]

At some point, you won't have to compile it yourself as it's possible to setup an autobuild system on Github unless you want
to add some funtions.

I've played with this on my own Fork that includes PlatformIO, Makefile and a Github autobuild system. Dustin has been notified with this progress.

Michael.

danielo515, 2021-12-09:
> At some point, you won't have to compile it yourself as it's possible to setup an autobuild system on Github unless you want
> to add some funtions.

Well, that is exactly my use case. So moving everything to the cloud is not, IMO, a good compromise to not make things more self contained?

closed 2022-06-05 by github-actions[bot]