DustinWatts/FreeTouchDeck #82: [FEATURE REQUEST] Deej compatibility

kingpin1905, 2022-01-08:
Add a Deej page/option

(From the deej github):
> Deej is an open-source hardware volume mixer for Windows and Linux PCs. It lets you use real-life sliders (like a DJ!) to seamlessly control the volumes of different apps (such as your music player, the game you're playing and your voice chat session) without having to stop what you're doing.

https://github.com/omriharel/deej
The Deej page could have some sort of faders.

briight, 2022-01-09:
iv had it working on both branches 

i think u have to add    BluetoothSerial SerialBT;     at the top but its been a long itme since i did it

DustinWatts, 2022-01-23:
What exactly do you mean by "Deej compatibility"?

kingpin1905, 2022-01-23:
> What exactly do you mean by "Deej compatibility"?

Add a page on the FreeTouchDeck and make the Arduino communicate with the Deej.exe (Arduino sends messages to the serial monitor and is picked up by the Deej.exe software)

closed 2022-06-05 by github-actions[bot]