DustinWatts/FreeTouchDeck #85: TFT shows white screen?

MoActive, 2022-04-08:
Hi Dustin,

I have setup everything as explained in the guides and keep getting a white screen.

Findings so far: 

- If I enable USECAPTOUCH the screen displays the icons, but are not clickable.
- if I leave everything as explained in the guide in FreeTouchDeck.h and only configure the user_setup.h file for driver ILI9488 the screen is just white.

Tried multiple things, but no luck?

Bought the same screen 3.5 SPI TFT ILI9488 Touch 320x480 as in the guide.

Thanks,

Mohammed

MoActive, 2022-04-08:
Finally figured out the issue.

Maybe this should be in the guide, but do not change the WIFI file at all. This was the problem.

closed 2022-04-08 by MoActive

DustinWatts, 2022-04-09:
Hi Mohammed,

>     If I enable USECAPTOUCH the screen displays the icons, but are not clickable.

If you use the screen from AliExpress it is a resistive touchscreen so do not uncomment the USECAPTOUCH line. 

>     Maybe this should be in the guide, but do not change the WIFI file at all. 

It could be that the ESP32 was in a continuous reset. You can check this by opening the serial monitor in Arduino at 115200 baud. If it keeps giving reset messages it is resetting before anything is drawn to the screen and the screen just goes white because of the backlight coming on.

Anyway, glad it works now! :)