DustinWatts/FreeTouchDeck #89: Add restart button to Wifi screen

ZainKergaye, 2022-05-11:
I think that there should be a restart button on your wifi screen just so that if it doesn't get configured correctly you don't have to manually pull the plug

DustinWatts, 2022-05-15:
Good idea!

DustinWatts, 2022-05-15:
Btw... FreeTouchDeck will always start and AP if it can't connect to your WiFi so you can always get to it, by connecting to the default AP.

Not-InstaGraham, 2022-05-19:
> Btw... FreeTouchDeck will always start and AP if it can't connect to your WiFi so you can always get to it, by connecting to the default AP. So maybe we leave the button out, because it always starts the webserver?

Seems like this would be a handy edition simply because of the sheer variety of setup variants users could have. Ensures a quick way to refresh the ESP without having to connect to the AP every time (assuming you’re having issues with the wifi).

DustinWatts, 2022-05-19:
Let's see if this works for everyone 😄 

Off course this is not in the release yet. So only manually compiling and uploading is needed.

DustinWatts, 2022-06-01:
No complains a of now. So closing. Feel free to reopen!

commit cc944c40 (2022-05-19): #89, LittleFS supoort, #90

- Renamed all SPIFFS to FILESYSTEM
- Users can now choose to use SPIFFS or LittleFS
- Fix #89 added a restart button when in WiFi mode
- Fix #90 Allow for multiple files uploads

Note! Because changes are made to the data folder, this version requires you to also upload the data folder again.
closed 2022-06-01 by DustinWatts