DustinWatts/FreeTouchDeck #93: Management of automated blinds

NigaMan, 2022-05-27:
And what if you add mqtt broker so that you can manage automated blinds?

DustinWatts, 2022-05-27:
There not enough RAM, so BLE, the TFT and WiFi can not be used at the same time. With a next version of the ESP32 TouchDown this might be possible. But not with this version of FreeTouchDeck.

closed 2022-05-27 by DustinWatts

Dark1886, 2024-08-06:
@DustinWatts This issue specifically mentioned an MQTT broker, but would it be possible to have the FreeTouchDeck just be an MQTT client that could execute some commands if it receives an MQTT message from an external broker?

DustinWatts, 2024-08-06:
It can be. But that would mean a really big re-write. I hope that in version 2.0. I can be either or even better, both. Especially when using the ESP32 S3 which is very powerful. I'm also thinking of making a board with the H series and the S3 which means there is more room for home automation. But for now, it is a no.

TobbeG, 2024-08-06:
@DustinWatts How are you doing now?