DustinWatts/FreeTouchDeck #95: Error when compiling FreetouchDeck-0.9.17

martinglass, 2022-07-02:
I get this error when compiling the latest release (FreetouchDeck-0.9.17). I'm using Arduino IDE, following the instructions in README.md

_**'KEY_NUM_ASTERISK' was not declared in this scope**_

This error relates to line 532-534 in Action.h
      case 11:
        bleKeyboard.write(KEY_NUM_ASTERISK);
        break;

DustinWatts, 2022-07-03:
Hi, this is because you are using an older version of the BLE Keyboard library where I spelled ASTERISK wrong (I spelled it asterix 😉 ) Try the latest version of the library: https://github.com/DustinWatts/ESP32-BLE-Keyboard

DustinWatts, 2022-08-17:
I assume this has been resolved? If not... feel free to reopen!

closed 2022-08-17 by DustinWatts