DustinWatts/FreeTouchDeck #97: Out of memory Error

martinglass, 2022-07-14:
I've appended your boards.txt content to the end of the Arduino boards.txt file, as described in your Wiki page (https://github.com/DustinWatts/esp32-touchdown/wiki/Uploading-code). The same thing happens when I use the ESP32-Dev board and the default settings shown in the Wiki.

I'm Verifying FreeTouchDeck v0.9.17 firmware in Arduino IDE v1.8.19 (Mac).

Using default setting (Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)) I get this out of memory error ...

**_Sketch uses 1642918 bytes (125%) of program storage space. Maximum is 1310720 bytes.text section exceeds available space in board

Global variables use 63200 bytes (19%) of dynamic memory, leaving 264480 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Error compiling for board ESP32 TouchDown_**

DustinWatts, 2022-07-14:
Hi Martin,

You should select "2MB APP/2MB SPIFFS". This resolves that issue. If not, feel free to reopen this issue!

martinglass, 2022-07-14:
> Hi Martin,
> 
> 
> 
> You should select "2MB APP/2MB SPIFFS". This resolves that issue. If not, feel free to reopen this issue! 
> 
> 
> 
> 

Thanks. Yes, changing the option worked, but I raised is an an Issue here more to signal an error in the Wiki. 

Maybe you'll change that?

DustinWatts, 2022-07-14:
Ah I see what you mean. That is for uploading code to the ESP32 TouchDown, not necessarily FreeTouchDeck....

martinglass, 2022-07-14:
> Ah I see what you mean. That is for uploading code to the ESP32 TouchDown, not necessarily FreeTouchDeck....

I'm using the ESP32 TouchDown though and uploading the FreeTouchDeck firmware to the ESP32 TouchDown.