DustinWatts/FreeTouchDeck #98: Work on WT32-SC01 3,5”?

Dansteris, 2022-08-16:
Hi,
I bought WT32-SC01 3,5” from 8ms, it board with ESP32 (ESP32 WROVER B) and LCD Capacitive touch, It will work on this board? Or you can add support for this board, its cheap and together ESP32 with LCD.
Thank you

DustinWatts, 2022-08-17:
I do not have the board myself, so a I can't confirm but try using this (Discord user `keefmac0843` TFT_eSPI User_Setup.h:

```
// User defined setup
#define ST7796_DRIVER

#define TFT_WIDTH  480
#define TFT_HEIGHT 320

#define TFT_BACKLIGHT_ON HIGH

#define USE_HSPI_PORT

#define TFT_MISO 1
#define TFT_MOSI 13
#define TFT_SCLK 14
#define TFT_CS   15
#define TFT_DC   21
#define TFT_RST  -1
#define TFT_BL   23
#define TOUCH_SDA 18
#define TOUCH_SCL 19
#define TOUCH_IRQ -1   ; not connected
#define TOUCH_RST -1
#define TOUCH_CS  22
```

manoper93, 2022-09-07:
Can you do .bin file flash or web installer for WT32-SC01?

DustinWatts, 2022-09-07:
I do not have a WT32-SC01 myself, so it would we difficult to test for me. I might get one and use it for testing. Until that time, you would have to compile for the WT32-SC01 yourself.

manoper93, 2022-09-07:
Please do that, my friends want too.
Can you send me est32devC.bin decompacted?

DustinWatts, 2022-09-07:
est32devC.bin is just the FreeTouchDeck.ino.

Dansteris, 2022-09-07:
Join to discord server, i can give you working config for WT32-SC01

manoper93, 2022-09-08:
In .ino i need to change touch driver to FT6336U, can you help me?

Dansteris, 2022-09-08:
Join to discord, and write to channel: #freetouchdeck, i will help you

closed 2022-11-15 by github-actions[bot]