DustinWatts/FreeTouchDeck #100: Buttons offset

hermannw, 2022-10-01:
I recently discovered this project and I love it! Thanks Dustin for sharing this!
I am using an ESP32DEVKITV1 and I used the web installer. Everything works great with two exceptions: I did the touch screen calibration multiple times but it looks like the buttons are off.  

[image]

This is roughly how the touch function works. So the touch function of the upper button row, starts about 2/3 from the upper edge. Any idea why this happens?
The other thing I had a few times, it that when the display goes to sleep for a long time (30 minutes or so), I cannot wake it up any more. Or am I missing something?
Hermann

hermannw, 2022-10-06:
Hi Dustin, yes I have redone the calibration several times and there is no change. I am using a veroboard with the ESP one one side and the LCD on the other. I tested it with a breadboard before and it showed the same offset. What could be wrong on the wiring? The ESP board is a different one but from what I understand, the main difference between them is the board and not the processor... I am planning to build another one so I am going to buy another LCD. Is there any possibility to define some offset in the software? I am familiar with the ESP programming. I just used the ready made image because I am lazy. :-)
Hermann

DustinWatts, 2022-10-07:
@hermannw The TFT touch calibration is a function directly from the [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI) library. The resistive touch functions are also taken directly from there. I always tried to fix the reason for the offset. I had it a lot of times during development, and it was always an error in the code. But in the images and the -master version here on Github no such errors exist as far as I know.

I forgot to ask, does the screen react to the touch when you do the calibration? I mean the two top corners.

Without knowing what ESP32 you are exactly you are using I can not say if it makes a difference. Chances are it does not because like you said, most ESP32 dev boards use the same WROOM module. But it could be that there is a difference on your board that makes SPI (resistive touch uses SPI) function differently.

I recommend compiling the code using the Arduino IDE. You can follow the instructions in the FreeTouchDeck.ino (the comments tell you what you need to know), on [Instructables](https://www.instructables.com/A-Bluetooth-ESP32-TFT-Touch-Macro-Keypad/), or here in the Wiki. I know this is a lot more work then using the image but you can compile for the board you are using. The images are compiled for the standard ESP32 DevKit.

hermannw, 2022-10-08:
The calibration works ok. I am using a stylus for better precision. All four corners react and at the end there is a message that the calibration was saved. 
I just downloaded the repository and compiled it. The result is the same. I'll try with another LCD. 
Thanks for the help!

hermannw, 2022-10-09:
Well, it was indeed the LCD... I replaced it with another one and that works.

closed 2022-10-09 by hermannw

DustinWatts, 2022-10-10:
> Is it possible that the restive network or the touch controller are screwed up?

Both are possible. But I am glad that it was the screen and that with another screen it works. I hope you can still get your money back for the faulty one. If not, you can replace only the touch film, it is difficult, but possible.