DustinWatts/FreeTouchDeck #101: Latching issue on screen 3 position 4

Tigorko, 2022-10-19:
Hello
First of all, thank you for your time developing this product, it's excellent and I love it and hope it will be even better in the future.
After two weeks of usage, I encountered a bug that does not go away even after a complete reinstall.

On Menu 3, position 4, latching does not work. I can press this button and the shortcut is executed, but the graphical update is not.

DustinWatts, 2022-10-19:
Hi @Tigorko Thanks for the report. I will look in to it and fix it!

DustinWatts, 2022-10-19:
Found a missing line in `ConfigLoad.h` line `970` should start with `menu3.button3.latch = doc["button3"]["latch"] | false;` will fix this in the next release.

closed 2023-02-25 by github-actions[bot]