DustinWatts/FreeTouchDeck #102: Getting just a partly lit screen

griffinmt, 2022-11-10:
Where are the pins of the ESP32 defined to work the Touchscreen, in particular T_CLK,T_DIN,T_OUT,T_IRQ. I have the screen wired up for T_CLK=G18,T_DIN=G23,T_OUT=G19,T_IRQ=G27

DustinWatts, 2022-11-10:
They are defined in User_Setup.h in the TFT_eSPI library. Also make sure you wire up the BL (Backlight) pin. Either to a pin defined in User_Setup.h or directly to 3.3V. If you hook it up to 3.3V the backlight is always lit so I do not recommend it, but it is a good way to test.

griffinmt, 2022-11-10:
But this touch pins are not defined anywhere. I would add them myself, but what are they named. For examble, is it TOUCH_IRQ or TFT_T_IRQ or just _IRQ.

DustinWatts, 2022-11-10:
They are not defined separately if you are using the ILI9488 module, becuase both the screen and touch use SPI. You can find the correct setup in for an ESP32 with ILI9488 module here: [ESP32_Dev_Kit_V1_ILI9488_Resistive.h](https://github.com/DustinWatts/FreeTouchDeck/blob/master/user_setup.h%20Examples/ESP32_Dev_Kit_V1_ILI9488_Resistive.h)

griffinmt, 2022-11-10:
So, I dont need to physically wire and pins to the display touch screen connections other than T_CS. Is that correct?

DustinWatts, 2022-11-11:
No, you also nee to wire T_DIN, T_DO, T_CS and T_CLK. Because although in software they are the same (except for T_DO, which is not used by the screen), the SPI pins on the module for the screen and the touch controller are not connected on the PCB.

griffinmt, 2022-11-11:
I was going to buy your pcb for doing this, as the first one was essentially free, till I found out there was a $29 shipping fee.

griffinmt, 2022-11-11:
Go some results today. I switched over to an ILI9341 display, and altered some pin assignments, and I got the button screen up!!! BUT that screen is smaller than the 88, so I only see the top two buttons and part of the two below them. Is there a way to shrink everything proportionally?

DustinWatts, 2022-11-11:
Yes there is. In FreeTouchDeck.ino you can define the height and width of the screen. 

[@ line 157](https://github.com/DustinWatts/FreeTouchDeck/blob/2960b802affec841c23b6c59b7a4a969813af89d/FreeTouchDeck.ino#L157)

```
#define SCREEN_WIDTH 480
#define SCREEN_HEIGHT 320
```

If you do that, everything should scale accordingly.

griffinmt, 2022-11-11:
I am a dummy some times!!!

DustinWatts, 2022-11-11:
If you've written the software, it is easy to find. 😉

griffinmt, 2022-11-11:
Just an FYI: These are my current settings that work on the 9341 but not on the 9488. Will order a new one.
Thanks for your help.
#define TFT_MISO 19
#define TOUCH_DO 19
#define TFT_MOSI 23
#define TOUCH_DIN 23
#define TFT_SCLK 18
#define TOUCH_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    27  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
#define TFT_BL   32  // LED back-light
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)
#define TFT_LED   32  // LED back-light
#define TOUCH_CS 21
#define TOUCH_IRQ 35

closed 2022-11-11 by griffinmt