DustinWatts/FreeTouchDeck #103: After web-install toch does not work

1wenter1, 2022-11-12:
Hi,
I have connected all the wires to the esp32 according to https://github.com/DustinWatts/ESP32_TFT_Combiner#these-are-the-connections-used-by-the-tft-and-sd-card
I have made all necessary changes in the tft config. I am using ILIU9488 screen.
After installation TouchDeck home screen is visible but touch itself does not work. I have tried manual installation, but after sketch upload it went to white screen. Is there a way to "define USECAPTOUCH" after web-install?

DustinWatts, 2022-11-12:
The ILI9488 module (the red one) with a separate ESP32 does not use capacitive touch but resistive touch. Are you sure you are installing the correct version with the webinstaller? And when you do it manually, you should keep USECAPTOUCH commented out.

1wenter1, 2022-12-01:
I have finally received the combiner pcb and soldered display and esp32. Everything works as intended now. 
Thank you

closed 2022-12-01 by 1wenter1

DustinWatts, 2022-12-01:
Good to hear @1wenter1 Thanks for letting me know and closing this issue, enjoy!