DustinWatts/FreeTouchDeck #108: Work on WT32-SC01 Plus

ranseyer, 2023-03-15:
Hi,

is somebody using or working on the use of an "WT32-SC01 Plus" Support ?

It is like this: https://github.com/DustinWatts/FreeTouchDeck/issues/98
But has an ESP32-S3.  Datasheet: https://www.antratek.de/media/wysiwyg/pdf/WT32-SC01-Plus-V1.3-EN.pdf

> Main controller of WT32-SC01 Plus development board is configured with a more integrated and powerful WT32-S3-Wrover-N16R2 module. this module is a general-purpose WiFi + BT + BLE MCU module, 16MB SPI flash and 2MB PSRAM configured in. It supports RGB colorful display of various pictures, capacitive multi-touch makes interaction easier, and the use of IPS full touch screen brings users better visual effects and a more comfortable experience.

Display Driver: ST7701
Touch-Driver: FT6336U via I2C

When i flash precompiled firmwares like Freetouchdeck / "ESP32-S3-DevKitC-1 (Using USB keyboard)" (flashed via the debug port) i get no picture on the screen and on the console:

> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x15 (USB_UART_CHIP_RESET),boot:0x2 (DOWNLOAD(USB/UART0))
>Saved PC:0x40041a79
> waiting for download

So i believe it is not enough to cahnge some pin assignments...

DustinWatts, 2023-03-15:
Hi @ranseyer on my Discord there are some people who ported FTD to the WT32-SC01. If you join and ask the question there you'll probably find the answers you are looking for.

closed 2023-03-15 by DustinWatts

ranseyer, 2023-03-16:
Thanks for the feedback. I asked the question in the discord.

But the WT32-SC01 and the WT32-SC01 **Plus** are complete different...
There are more topics than the pin assignments only

I would say: other types of
- ESP Chip
- RAM
- Display
- Touch

...

Source: https://www.antratek.de/media/wysiwyg/pdf/WT32-SC01-Plus-V1.3-EN.pdf