DustinWatts/FreeTouchDeck #113: Backlight brightness control not working?

DustinWatts, 2023-04-16:
Got a report that on the `-master` BL control does not work. Have to check.

DustinWatts, 2023-04-16:
I just checked and also on the ESP32 TouchDown S3 BL control isn't working. I have to investigate why not. Something goes wrong here: 

```
  ledcSetup(0, 5000, 8);
#ifdef TFT_BL
  ledcAttachPin(TFT_BL, 0);
#else
  ledcAttachPin(backlightPin, 0);
#endif // defined(TFT_BL)
  ledcWrite(0, ledBrightness); // Start @ initial Brightness
```

DustinWatts, 2023-04-16:
This is from 0.9.17, that is working:
```
  ledcSetup(0, 5000, 8);
#ifdef TFT_BL
  ledcAttachPin(TFT_BL, 0);
#else
  ledcAttachPin(32, 0);
#endif
  ledcWrite(0, ledBrightness); // Start @ initial Brightness
```

I have a feeling that it always skips `ifdef TFT_BL` and just attaches pin 32. Which on the ESP32 TouchDown as wel as on the DevKit version is the same and correct pin.

I made the changes to facilitate the BL control on the ESP32-S3 where pin 48 is used. I'm going to try with some Serial.print()'s to see what TFT_BL actually is.

XXGOD123, 2023-04-16:
> This is from 0.9.17, that is working:
>
> ```
>   ledcSetup(0, 5000, 8);
> #ifdef TFT_BL
>   ledcAttachPin(TFT_BL, 0);
> #else
>   ledcAttachPin(32, 0);
> #endif
>   ledcWrite(0, ledBrightness); // Start @ initial [...]

so, just change from ledcAttachPin(backlightPin, 0); to ledcAttachPin(32, 0);?

DustinWatts, 2023-04-16:
> so, just change from ledcAttachPin(backlightPin, 0); to ledcAttachPin(32, 0);?

Correct, this is a quick fix, but not a permanent fix for it to work across all boards.

XXGOD123, 2023-04-16:
> > so, just change from ledcAttachPin(backlightPin, 0); to ledcAttachPin(32, 0);?
> 
> Correct, this is a quick fix, but not a permanent fix for it to work across all boards.

What will happen if always use this method?

DustinWatts, 2023-04-16:
On the DevKit version as well as on the ESP32 TouchDown, nothing.

But, on the new (to be released, but not ETA) ESP32 TouchDown S3, the backlight pin is not 32 but 48. So I was trying to get the BL pin from the User_Setup.h of the the TFT_eSPI library. But this does not seem to work.

Raycast7373, 2023-04-16:
Backlight works fine for me (although the board i use isn't officially supported)

Raycast7373, 2023-04-16:
And i implemented auto brightness which runs in the main loop so there's that too

DustinWatts, 2023-04-16:
> Backlight works fine for me (although the board i use isn't officially supported)

Yes, if you define the correct pin for the brightness it should work.

closed 2023-06-22 by github-actions[bot]

OpossumTinte, 2024-06-18:
Hello, I jus came across this problem, but I don't know how to solve it. 

'ledcSetup' was not declared in this scope

this is in the same exact Code that you have posted. I would appreciate your help. 
Thank you in advance