DustinWatts/FreeTouchDeck #114: Compile issue

phpbbireland, 2023-08-08:
`FreeTouchDeck:128:10: fatal error: esp_bt_main.h: No such file or directory
 #include "esp_bt_main.h"   // Additional BLE functionaity
          ^~~~~~~~~~~~~~~
compilation terminated.
Using library SPIFFS at version 2.0.0 in folder: /home/Mike/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/SPIFFS 
Using library FS at version 2.0.0 in folder: /home/Mike/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/FS 
Using library Preferences at version 2.0.0 in folder: /home/Mike/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/Preferences 
Using library TFT_eSPI at version 2.5.0 in folder: /home/Mike/Arduino/libraries/TFT_eSPI 
Using library SPI at version 2.0.0 in folder: /home/Mike/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/SPI 
Using library USB at version 2.0.0 in folder: /home/Mike/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/USB 
Using library BLE at version 2.0.0 in folder: /home/Mike/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/BLE 
exit status 1
esp_bt_main.h: No such file or directory
`
Thank for the code...

DustinWatts, 2023-08-08:
`esp_bt_main.h` is removed from the latest ESP32 Arduino Core. I have no idea why. A quick solution would be downgrading the ESP32 board core you installed to something lower then 2.0.11. Maybe 10 or even 9. 

But I suspect to see this error a lot more the coming time for people installing a fresh FreeTouchDeck. So I have to see where the moved the functions that where in `esp_bt_main.h`. And fix the FreeTouchDeck code.

These 4 functions are from the additional included files:

```
btStop();
esp_bt_controller_disable();
esp_bt_controller_deinit();
esp_bt_controller_mem_release(ESP_BT_MODE_BTDM);
```

So as long as we have those... we are ok. Maybe you could try also to comment out `#include esp_bt_main.h` and hope that those functions are included in another way. If works it also makes my code fix easier :)

DustinWatts, 2023-08-08:
Just to be sure... you are using an ESP32 DevKit for the original error right? FreeTouchDeck is only for an ESP32.

phpbbireland, 2023-08-08:
I was convinced it was for the ESP32, but it must have been the ESP32S2 I selected, as I can only repeat it for this version.

In the mean time I ordered 5 of your PCB's from PCBWay and an a couple of the ESP-WROOM-32D so everything matches your build... 

Apologies if I screwed up and reported an error where none existed, I'm blaming old age for it. In the mean time I will try again with your suggestions... Mike

DustinWatts, 2023-08-08:
So if you use the board it was intended for there is no error? No problem is this is the case. I should have checked :D 

I assume if you select the ESP32 Devkit it works as intended?

phpbbireland, 2023-08-08:
Update... All good, compiled fine now to wire it up...

DustinWatts, 2023-08-08:
Great! I will close the issue. May you run in to another issue feel free to open another one! 😄

closed 2023-08-08 by DustinWatts