DustinWatts/FreeTouchDeck #115: Request

phpbbireland, 2023-09-12:
Will the code work with Makerfabs-ESP32-C3-SPI-TFT-with-Touch boards?

Mike

DustinWatts, 2023-09-17:
Hi @phpbbireland!

FreetouchDeck should work on any ESP32 that has WiFi and BLE. I haven't tested it. But you should be able to change the User_Setup.h for your screen and board. And also change some #defines in the main *.ino sketch. Let me know if it works out!

phpbbireland, 2023-09-19:
Thanks, will try some more...

DustinWatts, 2023-10-15:
This issue is still open... did you make any progress?

closed 2023-12-21 by github-actions[bot]

phpbbireland, 2023-12-24:
Sorry, real busy, forgot to close this...
Haven't got a chance to test but the extra info should  suffice...

DustinWatts, 2023-12-24:
You can always reopen this issue if you make some progress and show it to other interested in doing the same thing! 👍