DustinWatts/FreeTouchDeck #123: ESP32-2432S028R (Cheap Yellow Display) support

killabeenl, 2024-02-20:
Is it possible to add the ESP32-2432S028R (Cheap Yellow Display) to the web installer?
It is a nice all in one display with an esp32. But the display won't work with the standard installer

DustinWatts, 2024-02-21:
Hey Killabee. I know the display. But the code for it is not mine. So I can just "steal" the binary file for the web installer. I'll have to talk to the person who forked the original library.

What code are you talking about?

killabeenl, 2024-02-21:
Hello Dustin,

Thanks for the response. I tried to flash the device with: https://install.freetouchdeck.com/
I choose the ESP32 DevKitC option en flash the device. That is working.
But with the flash is complete the display does not work.

I also tried the manual install with the Arduino IDE software but got some errors there. I did some reading in your Discord channel and found that I need an older version of Arduino IDE.

Will try a manual installation tonight.

DustinWatts, 2024-02-21:
I thought there was fork out there... but can't find it.

killabeenl, 2024-02-22:
The manual compilation did not work. Got a error that I need to investigate.
I will close the case and discuss this in your discord channel if needed.

closed 2024-02-22 by killabeenl