DustinWatts/FreeTouchDeck #124: https://install.freetouchdeck.com/ is down

becdetat, 2024-05-17:
Hi, I tried to use https://install.freetouchdeck.com/ however the site appears to be down.

DustinWatts, 2024-05-17:
Hi Rebecca,

I know. It is unfortunate. I am working on a solution. But it means that I have to move the whole `install.*.*` subdomain to a new hosting platform. I am thinking Github pages and I am figuring out how to do this properly. But... it will return! 

In the mean time if there is anything I can do, please let me know. Also, thank you for making this a real issue. I need a kick in the but! :)

DustinWatts, 2024-05-22:
Hi Rebecca,

It should be fixed now. In the README.md file of the link to the working installer has been changed. The new working URL is: https://dustinwatts.github.io/install-freetouchdeck/

If you still have issues, please feel free to re-open this issue. But it should be ok now!

closed 2024-05-22 by DustinWatts

becdetat, 2024-05-22:
Thanks @DustinWatts :-) I managed to flash my device. Not successfully unfortunately, but I'll open another issue for that.